stm32duino/STM32Ethernet #4: Ethernet.begin() does not return under certian circumstances

pulli7, 2017-09-22:
In the past, with the Arduino-Ethernet-Shield, I used the following code to wait for ethernet connection to get available, so that I can have at least some limited functionality when ethernet is not available at this time, and automatically jump to full functionality as soon as link gets available.
```
void setup()
{
  if (Ethernet.begin(mac) == 0)
  {
    FallBackMode();
  } 
}
```

```
void FallBackMode()
{
  int Timestamp = millis();
  while(1)
  {
     if ( millis() - Timestamp > 300000)  //try reconnect every 5min
     {
       if (Ethernet.begin(mac, 30000) != 0)  //Problem here
       {                                   
         break;  //successfully connected
       }
       Timestamp = millis();
    }
    //call  "offline application"  here, which serves all functionality that does not need Ethernet
  }//ENDwhile
}
```

With the Nucleo 429ZI this does not work. As long as the cable is unplugged everthing is fine, but as soon as it is plugged in and ethernet gets available, the program gets locked inside the Ethernet.begin() function and never returns.

Am I misusing the function and it worked just coincidentally with the standard ethernet-shield, or is this a bug?
If this usage of the begin() function is not intended, is there another way to achieve that the board tries reconnecting itself?

fprwi6labs, 2017-09-26:
Issue reproduced. This is under investigation.

fpistm, 2017-10-18:
Hi all,
maybe the PR #5 fix this issue as the Ethernet scheduler is called by timer every 1ms?

pulli7, 2017-10-18:
Gave it a quick try:

Looks like it changes the behavior in the way, that now calling Ethernet.begin() a second time, after the first try failed, always locks the program inside the function, no matter if ethernet is still unavailable or not.

fprwi6labs, 2017-10-19:
@pulli7 
Could you try the PR #5? It fixes you issue with several call of begin().

pulli7, 2017-10-20:
Confirmed, issue is fixed.
Thank you again @fprwi6labs !

fprwi6labs, 2017-10-20:
Your welcome @pulli7 .
@fpistm you can close all those issues.

fpistm, 2017-10-20:
Thanks guy.

closed 2017-10-20 by fpistm

fpistm, 2018-04-25:
This library is designed for STM32 boards with native Ethernet.
You should submit your issue on https://github.com/adafruit/Ethernet2 or https://github.com/arduino/Arduino/tree/master/libraries/Ethernet depending of your shield.