stm32duino/STM32Ethernet #6: stm32_eth.c unknown type name 'stimer_t'

gurulator, 2017-10-30:
I am trying to compile ethernet examples for F746-discovery, but there is always the same error shown the picture (Documents\Arduino\libraries\STM32Ethernet-master\src\utility\stm32_eth.c:100:8: error: unknown type name 'stimer_t'). I already installed LwIP-master. I have no idea what's wrong.
[image]

fpistm, 2017-10-31:
Hi @gurulator ,
Which core you used?

gurulator, 2017-10-31:
ARM® Cortex®-M7 core-based STM32F746NGH6 microcontroller, and the STM32GENERIC  implementation. https://github.com/ChrisMicro/STM32GENERIC
All the example codes are working well incuded in STM32GENERIC except the ethernet examples.
I am able to use the I/O pins and the TFT capacitive screen, but when I include the <STM32Ethernet.h>, the comiler displays the error report shown above.
http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-discovery-kits/32f746gdiscovery.html

fpistm, 2017-10-31:
Ok that's why I thought. The latest version of this library is not compatible with STM32generic core.
This library is mainly for the https://github.com/stm32duino/Arduino_Core_STM32
Library now use a timer to call periodically the lwip scheduler.
Ask to @danieleff if he wants add wrapper to timer use in his core, here:
https://github.com/danieleff/STM32GENERIC/tree/master/STM32/libraries/OfficialCompatibility

gurulator, 2017-10-31:
Ok, thanks for help.

gurulator, 2017-10-31:
Now i tried it with the First release and it works fine, that was the problem.
https://github.com/stm32duino/STM32Ethernet/releases/tag/1.0.0
Thanks one more time!

fpistm, 2017-10-31:
No worries ;)
It should be fine to unified API.
It's planned to extend timer API for https://github.com/stm32duino/Arduino_Core_STM32
We will try to do it as generic as possible.

closed 2017-10-31 by fpistm