stm32duino/STM32Ethernet #9: MAC address

sstaub, 2018-03-31:
The build in mac address should used. Following function can used, based on [https://github.com/ARMmbed/mbed-os/blob/75537a89bd05cfe090a89059bfafc0f8a2b73802/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/stm32xx_emac.c](url)
`void stmMAC(uint8_t *mac) {
  // Read unic id
  #if defined (STM32F2)
    uint32_t baseUID = *(uint32_t *)0x1FFF7A10;
  #elif defined (STM32F4)
    uint32_t baseUID = *(uint32_t *)0x1FFF7A10;
  #elif defined (STM32F7)
    uint32_t baseUID = *(uint32_t *)0x1FF0F420;
  #else
    #error MAC address can not be derived from target unique Id
  #endif
  
  mac[0] = 0x00;
  mac[1] = 0x80;
  mac[2] = 0xE1;
  mac[3] = (baseUID & 0x00FF0000) >> 16;
  mac[4] = (baseUID & 0x0000FF00) >> 8;
  mac[5] = (baseUID & 0x000000FF);
  return;
}`

fpistm, 2018-03-31:
Hi @sstaub ,
good idea.
Do not hesitate to submit a PR for that or any other contributor ;)

sstaub, 2018-06-01:
I did make a PR #11

closed 2018-06-01 by sstaub