stm32duino/STM32Ethernet #14: is it having SSL

Rajkumar181, 2018-10-30:


fpistm, 2018-10-30:
Hi @Rajkumar181 
there is no ssl support.

Rajkumar181, 2018-10-30:
Any other way or library to add ssl on it.

fpistm, 2018-10-30:
This could probably be done  but I never investigate this.

Rajkumar181, 2018-11-14:
if i took the library for STM32CubeMx . so how can i attach to it.

fpistm, 2018-11-14:
I guess you talk about mbedtls?
Maybe the best way would be to convert it as an Arduino Library to be able to build it.
Then update the STM32Ethernet to properly use it.
This is just guess, I do not made any study/test and I must admit I do not know mbedtls.

Rajkumar181, 2018-11-14:
yes you are right i need mbedtls on Ethernet.

gdsports, 2019-10-16:
I am working on adding STM32Ethernet support to this library. It currently supports SAMD and Wiznet Ethernet.

https://github.com/OPEnSLab-OSU/SSLClient

gdsports, 2019-10-18:
Here is my fork with experimental support for the Nucleo F767ZI board.

https://github.com/gdsports/SSLClient/tree/stm32ethernet

The EthernetHTTPS example works.

The EthernetMultiHTTPS example does not. The example works with W5x00 Ethernet so there is some difference in STM32 Ethernet that I do not understand.

I added the howsmyssl (https://www.howsmyssl.com/) example that shows the supported cipher suites, TLS version, and an overall rating. All looks good.

```
  {
  "given_cipher_suites": [
    "TLS_ECDHE_ECDSA_WITH_CHACHA20_POLY1305_SHA256",
    "TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256",
    "TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256",
    "TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256",
    "TLS_ECDH_ECDSA_WITH_AES_128_GCM_SHA256",
    "TLS_ECDH_RSA_WITH_AES_128_GCM_SHA256"
  ],
  "ephemeral_keys_supported": true,
  "session_ticket_supported": false,
  "tls_compression_supported": false,
  "unknown_cipher_suite_supported": false,
  "beast_vuln": false,
  "able_to_detect_n_minus_one_splitting": false,
  "insecure_cipher_suites": {},
  "tls_version": "TLS 1.2",
  "rating": "Probably Okay"
  }
```

Lots of room for more cipher suites. 

```
Sketch uses 117252 bytes (5%) of program storage space. Maximum is 2097152 bytes.
Global variables use 49536 bytes (9%) of dynamic memory, leaving 474752 bytes for local variables. Maximum is 524288 bytes.
```

The TLS connect time (key exchange and certificate chain authentication) is about 860 ms versus 2850 ms on the SAMD21 Cortex M0 @ 48 MHz.

fpistm, 2020-02-13:
Close as answered.
All required stuff required in this library merged. thanks @gdsports .
Support addition on going here:
https://github.com/OPEnSLab-OSU/SSLClient/issues/3

closed 2020-02-13 by fpistm