stm32duino/STM32Ethernet #15: Support for DP83848_PHY ?

stevstrong, 2018-11-14:
Hi,
tried to use this lib on a Arch Max board which is equipped with DP83848C chip, but compile fails for ethernetif.c on accessing register PHY_IMR, which is obviously only defined for LAN8742A_PHY by this lib.

Searching the Arduino_Core_STM32 files, I could find in several headers `DP83848_PHY_ADDRESS` and its registers. However, PHY_IMR is not defined there. What is then the reason to define this phy at all in the core? The STM32Ethernet lib seems not supporting this PHY. How can this PHY be used with ST core?

Is there any plan for this lib to support this PHY?

fpistm, 2018-11-16:
Hi @stevstrong 
`DP83848_PHY_ADDRESS`  is in fact by default in the `stm32f4xx_hal_conf.h` because this is the default in the `stm32f4xx_hal_conf_template.h`.
Yes, I never test with this as I do not test a board with this.
This could be fine to do that.

fpistm, 2019-11-10:
Track in #34

closed 2019-11-10 by fpistm