stm32duino/STM32Ethernet #18: New init procedure

sstaub, 2019-01-31:
The new init procedure, using the build in mac Address doesn't work.
The problem is that overloading doesn't work because "IPAddress local_IP" and "uint8_t * mac" are recognized as the same type. So you have decide which type procedure should we use, the stupid old style or my new style, both at the same time is not possible. When using the old style I can make the macAddressDefault(void) function as a public type.

fpistm, 2019-01-31:
Strange as IPAddress  is a class.
I will have to check that. Thanks for report.

sstaub, 2019-02-04:
Found the problem:
uint8_t ip[] = {10, 101, 1, 201}; doesn't work
IPAddress ip(10, 101, 1, 201); this works
For init you must take the class IPAddress.

fpistm, 2019-02-17:
@sstaub this one could be close?

sstaub, 2019-02-17:
yes

closed 2019-02-17 by sstaub