stm32duino/STM32Ethernet #21: Ethernet library compilation error

KOUKILEO, 2019-06-03:
I am using the stm32 Ethernet network library.
But he told me that the compilation error.
I am a newcomer to stm32.
How can I solve him ? thank .
The development board I am using is STM32 Nucleo-144.

[image]

fpistm, 2019-06-03:
Which board exactly?
There is several Nucleo144.

KOUKILEO, 2019-06-03:
> 哪個板完全正確？
> 有幾個Nucleo144。

Is stm32 nucleo-h743zi
https://os.mbed.com/platforms/ST-Nucleo-H743ZI/

fpistm, 2019-06-03:
Ok thanks for the feedback.
Currently, this is a know issue.
Ethernet on H7 is not managed like other platform.
When I've introduced the H7 series, I've point it out in the restriction:
https://github.com/stm32duino/Arduino_Core_STM32/pull/392

KOUKILEO, 2019-06-03:
> 好的，謝謝你的反饋。
> 目前，這是一個已知的問題。
> H7上的以太網不像其他平台那樣管理。
> 當我介紹H7系列時，我在限制中指出了它：
> [stm32duino / Arduino_Core_STM32＃392](https://github.com/stm32duino/Arduino_Core_STM32/pull/392)

Ok, thank you for your help! I look forward to being supported as soon as possible.

OmarAlkassab, 2020-05-11:
Any news for the Support of Nucleo-H743 board?

fpistm, 2020-05-11:
Hi @OmarAlkassab 
Currently, I had no time to work on this.
Hope to have some time soon to work on this.

OmarAlkassab, 2020-05-11:
Many thanks dear

OmarAlkassab, 2020-05-11:
Dear @fpistm
I am newbie with STM32 microcontrollers, and I need to run a small web-server using the Ethernet port on the Nucleo board, can you suggest me how to easily run it?

markmaker, 2020-11-16:
Plus one. ;-)

sstaub, 2021-02-27:
Any news about?

fpistm, 2021-05-28:
> 
> 
> Any news about or some hints for solving the problem?

Unfortunately not. Currently, no time to work on this. Sorry.

fpistm, 2021-10-20:
STM32H7 Ethernet part diverge from other, so it will request to integrate it properly and also handle properly each PHY (#34)
This will required some time as currently I've no free slot, sorry.

LevitateGamer, 2022-05-28:
Do you have any updates waiting now/being worked on? I have the stm32h743zi2

> STM32H7 Ethernet part diverge from other, so it will request to integrate it properly and also handle properly each PHY (#34) This will required some time as currently I've no free slot, sorry.

EricZeiberg, 2022-07-29:
Any update on this? If not, is there a workaround in the meantime?

fpistm, 2023-03-20:
Unfortunately, time is always the missing ingredient. This is a community project, any contribution are welcome. It is always in the pipe but can't say when.

4ndreas, 2023-03-22:
> Unfortunately, time is always the missing ingredient. This is a community project, any contribution are welcome. It is always in the pipe but can't say when.

understandable, It took me 3 days but I managed to get it working.
https://github.com/4ndreas/STM32Ethernet
I added an example with TCP and UDP that I used, everything else is untested. 

however there are so many changes that it is hard to integrate in the current version.

The biggest problem is the addition to the linker-script and the MPU_Config to unlock the memory.
I think the best way would be to create a variant of the board with and without ethernet?

fpistm, 2023-03-23:
Hi @4ndreas 
Thanks for sharing.
About linker script we could probably use preprocessing to enable or not the dedicated part as the linker file is preprocessed.
About the integration, as mentioned above, the generic phy support should be implemented, I guess.

fpistm, 2023-03-23:
It refers to this: https://github.com/stm32duino/STM32Ethernet/issues/34
As it would requires some rework it should be done first.

fpistm, 2023-11-07:
Will be handled by #34

closed 2023-11-07 by fpistm

Uday2024, 2023-11-18:
Hi @fpistm , now the stm32h743zi MCU supports the STM32 ethernet right?

fpistm, 2023-11-18:
> Hi @fpistm , now the stm32h743zi MCU supports the STM32 ethernet right?

No. As stated in my previous comment:

> Will be handled by #34