stm32duino/STM32Ethernet #23: Compilation error occurs on Generic STM32F4 series

nopnop2002, 2019-07-24:
I'm try to compile [this](https://github.com/stm32duino/STM32Ethernet/tree/master/examples/UdpNtpClient) on Generic STM32F4 Series.

I installed these.

https://github.com/stm32duino/STM32Ethernet

https://github.com/stm32duino/LwIP

[image]

But compile error occure.
This is compile options.

[image]

Do you have any advice.

		Warning: Board atmega64:avr:memduino64_8MHz doesn't define a 'build.board' preference. Auto-set to: AVR_MEMDUINO64_8MHZ
		Warning: Board atmega644_1284:avr:avrmega644p doesn't define a 'build.board' preference. Auto-set to: AVR_AVRMEGA644P
		Warning: Board atmega644_1284:avr:avrmega644p16 doesn't define a 'build.board' preference. Auto-set to: AVR_AVRMEGA644P16
		Warning: Board atmega644_1284:avr:avrmega1284p doesn't define a 'build.board' preference. Auto-set to: AVR_AVRMEGA1284P
		Warning: Board atmega644_1284:avr:avrmega1284p16 doesn't define a 'build.board' preference. Auto-set to: AVR_AVRMEGA1284P16
		�T�u�f�B���N�g���܂��̓t�@�C�� sketch �͊��ɑ��݂��܂��B

		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:74:15: error: unknown type name 'ETH_DMADescTypeDef'; did you mean 'RTC_DateTypeDef'?
		 __ALIGN_BEGIN ETH_DMADescTypeDef  DMARxDscrTab[ETH_RXBUFNB] __ALIGN_END;/* Ethernet Rx MA Descriptor */
		               ^~~~~~~~~~~~~~~~~~
		               RTC_DateTypeDef
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:79:15: error: unknown type name 'ETH_DMADescTypeDef'; did you mean 'RTC_DateTypeDef'?
		 __ALIGN_BEGIN ETH_DMADescTypeDef  DMATxDscrTab[ETH_TXBUFNB] __ALIGN_END;/* Ethernet Tx DMA Descriptor */
		               ^~~~~~~~~~~~~~~~~~
		               RTC_DateTypeDef
		In file included from C:\Users\user\Documents\ArduinoData\packages\STM32\hardware\stm32\1.6.1\system/STM32F4xx/stm32f4xx_hal_conf.h:13,
		                 from C:\Users\user\Documents\ArduinoData\packages\STM32\hardware\stm32\1.6.1\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:30,
		                 from C:\Users\user\Documents\ArduinoData\packages\STM32\hardware\stm32\1.6.1\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h:250,
		                 from C:\Users\user\Documents\ArduinoData\packages\STM32\hardware\stm32\1.6.1\cores\arduino/stm32/stm32_def.h:35,
		                 from \\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:48:
		C:\Users\user\Documents\ArduinoData\packages\STM32\hardware\stm32\1.6.1\system/STM32F4xx/stm32f4xx_hal_conf_default.h:227:40: error: 'ETH_MAX_PACKET_SIZE' undeclared here (not in a function); did you mean 'ETH_TX_BUF_SIZE'?
		 #define ETH_RX_BUF_SIZE                ETH_MAX_PACKET_SIZE /* buffer size for receive               */
		                                        ^~~~~~~~~~~~~~~~~~~
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:84:44: note: in expansion of macro 'ETH_RX_BUF_SIZE'
		 __ALIGN_BEGIN uint8_t Rx_Buff[ETH_RXBUFNB][ETH_RX_BUF_SIZE] __ALIGN_END; /* Ethernet Receive Buffer */
		                                            ^~~~~~~~~~~~~~~
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:91:8: error: unknown type name 'ETH_HandleTypeDef'
		 static ETH_HandleTypeDef EthHandle;
		        ^~~~~~~~~~~~~~~~~
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:105:22: error: unknown type name 'ETH_HandleTypeDef'; did you mean 'RTC_HandleTypeDef'?
		 void HAL_ETH_MspInit(ETH_HandleTypeDef *heth)
		                      ^~~~~~~~~~~~~~~~~
		                      RTC_HandleTypeDef
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c: In function 'low_level_init':
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:157:12: error: request for member 'Instance' in something not a structure or union
		   EthHandle.Instance = ETH;
		            ^
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:158:12: error: request for member 'Init' in something not a structure or union
		   EthHandle.Init.MACAddr = macaddress;
		            ^
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:159:12: error: request for member 'Init' in something not a structure or union
		   EthHandle.Init.AutoNegotiation = ETH_AUTONEGOTIATION_ENABLE;
		            ^
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:159:36: error: 'ETH_AUTONEGOTIATION_ENABLE' undeclared (first use in this function); did you mean 'PHY_AUTONEGOTIATION'?
		   EthHandle.Init.AutoNegotiation = ETH_AUTONEGOTIATION_ENABLE;
		                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~
		                                    PHY_AUTONEGOTIATION
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:159:36: note: each undeclared identifier is reported only once for each function it appears in
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:160:12: error: request for member 'Init' in something not a structure or union
		   EthHandle.Init.Speed = ETH_SPEED_100M;
		            ^
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:160:26: error: 'ETH_SPEED_100M' undeclared (first use in this function); did you mean 'I2C_SPEED_FAST'?
		   EthHandle.Init.Speed = ETH_SPEED_100M;
[... 154 lines cut ...]
		                  ^
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:560:32: error: 'ETH_SPEED_10M' undeclared (first use in this function)
		         EthHandle.Init.Speed = ETH_SPEED_10M;
		                                ^~~~~~~~~~~~~
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:565:18: error: request for member 'Init' in something not a structure or union
		         EthHandle.Init.Speed = ETH_SPEED_100M;
		                  ^
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:565:32: error: 'ETH_SPEED_100M' undeclared (first use in this function); did you mean 'I2C_SPEED_FAST'?
		         EthHandle.Init.Speed = ETH_SPEED_100M;
		                                ^~~~~~~~~~~~~~
		                                I2C_SPEED_FAST
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:576:74: error: request for member 'Init' in something not a structure or union
		       HAL_ETH_WritePHYRegister(&EthHandle, PHY_BCR, ((uint16_t)(EthHandle.Init.DuplexMode >> 3) |
		                                                                          ^
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:577:74: error: request for member 'Init' in something not a structure or union
		                                                      (uint16_t)(EthHandle.Init.Speed >> 1)));
		                                                                          ^
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:581:36: error: 'ETH_MACInitTypeDef' undeclared (first use in this function); did you mean 'RTC_InitTypeDef'?
		     HAL_ETH_ConfigMAC(&EthHandle, (ETH_MACInitTypeDef *) NULL);
		                                    ^~~~~~~~~~~~~~~~~~
		                                    RTC_InitTypeDef
		\\LANDISK-HDL-AA1\disk1\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.c:581:56: error: expected expression before ')' token
		     HAL_ETH_ConfigMAC(&EthHandle, (ETH_MACInitTypeDef *) NULL);
		                                                        ^
		exit status 1

fpistm, 2019-07-24:
Hi @nopnop2002 
the issue is that the `HAL_ETH_MODULE_ENABLED` is not enabled by default for this board (I guess you use the DIYMROE_F407VGT.
You can enable it by adding a `hal_conf_extra.h` file at sketch level with:
`#define HAL_ETH_MODULE_ENABLED`

nopnop2002, 2019-07-24:
Yes.
I'm using DIYMROE_F407VGT.

I tried 2 pattern.

Case #1
Add hal_conf_extra.h TAB.

[image]

Case #2
Add hal_conf_extra.h file in sketch folder.

[image]

But both case occure these error:

	Warning: Board atmega644_1284:avr:avrmega644p doesn't define a 'build.board' preference. Auto-set to: AVR_AVRMEGA644P
	Warning: Board atmega644_1284:avr:avrmega644p16 doesn't define a 'build.board' preference. Auto-set to: AVR_AVRMEGA644P16
	Warning: Board atmega644_1284:avr:avrmega1284p doesn't define a 'build.board' preference. Auto-set to: AVR_AVRMEGA1284P
	Warning: Board atmega644_1284:avr:avrmega1284p16 doesn't define a 'build.board' preference. Auto-set to: AVR_AVRMEGA1284P16
	Warning: Board atmega64:avr:memduino64_8MHz doesn't define a 'build.board' preference. Auto-set to: AVR_MEMDUINO64_8MHZ
	�T�u�f�B���N�g���܂��̓t�@�C�� sketch �͊��ɑ��݂��܂��B

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\STM32Ethernet-master\utility\ethernetif.c.o: in function `low_level_output':

	ethernetif.c:(.text.low_level_output+0x1e): undefined reference to `HAL_ETH_TransmitFrame'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\STM32Ethernet-master\utility\ethernetif.c.o: in function `ethernetif_input':

	ethernetif.c:(.text.ethernetif_input+0xa): undefined reference to `HAL_ETH_GetReceivedFrame_IT'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\STM32Ethernet-master\utility\ethernetif.c.o: in function `ethernetif_init':

	ethernetif.c:(.text.ethernetif_init+0x3a): undefined reference to `HAL_ETH_Init'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_init+0x54): undefined reference to `HAL_ETH_DMATxDescListInit'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_init+0x60): undefined reference to `HAL_ETH_DMARxDescListInit'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_init+0xa2): undefined reference to `HAL_ETH_Start'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_init+0xac): undefined reference to `HAL_ETH_ReadPHYRegister'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_init+0xbc): undefined reference to `HAL_ETH_WritePHYRegister'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\STM32Ethernet-master\utility\ethernetif.c.o: in function `ethernetif_set_link':

	ethernetif.c:(.text.ethernetif_set_link+0xe): undefined reference to `HAL_ETH_ReadPHYRegister'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_set_link+0x24): undefined reference to `HAL_ETH_ReadPHYRegister'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: libraries\STM32Ethernet-master\utility\ethernetif.c.o: in function `ethernetif_update_config':

	ethernetif.c:(.text.ethernetif_update_config+0x20): undefined reference to `HAL_ETH_WritePHYRegister'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_update_config+0x32): undefined reference to `HAL_ETH_ReadPHYRegister'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_update_config+0x50): undefined reference to `HAL_ETH_ReadPHYRegister'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_update_config+0x74): undefined reference to `HAL_ETH_ConfigMAC'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_update_config+0x7a): undefined reference to `HAL_ETH_Start'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_update_config+0x98): undefined reference to `HAL_ETH_WritePHYRegister'

	c:/users/user/documents/arduinodata/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: ethernetif.c:(.text.ethernetif_update_config+0xa0): undefined reference to `HAL_ETH_Stop'

	collect2.exe: error: ld returned 1 exit status
	exit status 1

fpistm, 2019-07-24:
Right you have to restart because arduino does not see change....
So this is ok now?

nopnop2002, 2019-07-24:
I'm using LAN8720,
I referred this page.

https://os.mbed.com/users/hudakz/code/STM32F407VET6_Hello/shortlog/

In Ethernet.begin, it repeat Link Up.
And finally become failed.

[image]

fpistm, 2019-07-25:
Well hard to tell what's going wrong.
I never test with this module, will try to find one but don't know when.
You will have to try to debug on your side.

nopnop2002, 2019-07-25:
Hello.

The problem has not been solved yet, but I will close this issue.
What do you think? ?

fpistm, 2019-07-25:
Original issue is solved.
Currently this Library support LAN8742A so mabe some update required to support the LAN8720.
You can try to see if the PHY configuration section is aligned with this:
https://github.com/stm32duino/Arduino_Core_STM32/blob/85fd492c15a87048086e7e82318c555fb6410a41/system/STM32F4xx/stm32f4xx_hal_conf_default.h#L233-L273

nopnop2002, 2019-07-26:
I changed stm32f4xx_hal_conf_default.h to this:
It work fine!!.

	/* Section 2: PHY configuration section */
	/* LAN8742A PHY Address*/
	//#define LAN8742A_PHY_ADDRESS            0x00U
	#define LAN8742A_PHY_ADDRESS            0x01U

[image]

Eventually I changed to this:

# stm32f4xx_hal_conf_default.h file 
at Documents\ArduinoData\packages\STM32\hardware\stm32\1.6.1\system\STM32F4xx

	/* Section 2: PHY configuration section */
	#if !defined  (LAN8742A_PHY_ADDRESS)
	/* LAN8742A PHY Address*/
	#define LAN8742A_PHY_ADDRESS            0x00U
	#endif

# hal_conf_extra.h file 
at sketch level

	#define HAL_ETH_MODULE_ENABLED
	#define LAN8742A_PHY_ADDRESS 0x01U

I'll pull request.

Thank you for your help.

fpistm, 2019-11-10:
Track in #34

closed 2019-11-10 by fpistm