stm32duino/STM32Ethernet #32: Compile Error with STM32F407VET

andy-hopkins, 2019-11-07:
Hi all,

I've just downloaded and installed into my Arduino 1.8.9 IDE, t
[Errors.txt](https://github.com/stm32duino/STM32Ethernet/files/3817406/Errors.txt)
he Arduino_Core_STM32 per the "Getting Started Instructions", and the STM32Ethernet library.

I'm using an STM32F407VET6 Black, so  I've selected the "Generic STM32F4 Series" board and the "Black 407VE" sub-option (also tried the FK407M1 SMT32F407VET and DIYMORE SMT32F407VGT with the same results)

Next, I've opened the ChatServer example program. When I click verify and it gets as far as "Compiling library "STM32duino_STM32Ethernet"", the errors the attached "Errors.txt" come up

I've also implemented the solution here ... https://github.com/stm32duino/STM32Ethernet/issues/23#issuecomment-514665499 
... to overcome a bunch of other errors.

I'm still pretty green with C/C++ so I'm a bit reluctant to dive in and start changing things like stimer_t to timer_t, I have no idea of the implications.

Can anyone throw any light on my issue.

fpistm, 2019-11-07:
Hi @andy-hopkins 
Right, this is normal. In fact I do not release yet the updated version of the lib for the new HardwareTimer implementation. To fix that, download the repo as a zip and install it with Arduino IDE.
This will fix your issue.

oguzhandalgic, 2019-11-07:
Hi @fpistm I was facing the same issue with @andy-hopkins and tried the method you have recommended yet now there is another error,

/Documents/Arduino/libraries/STM32Ethernet-master/src/utility/stm32_eth.h:47:10: fatal error: lwip/ip_addr.h: No such file or directory
 #include "lwip/ip_addr.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.

fpistm, 2019-11-07:
You need to install LwIP library... as described here:
https://github.com/stm32duino/STM32Ethernet#dependency

oguzhandalgic, 2019-11-07:
It is already installed, I installed it using the library manager of Arduino IDE

andy-hopkins, 2019-11-08:
> Hi @andy-hopkins
> Right, this is normal. In fact I do not release yet the updated version of the lib for the new HardwareTimer implementation. To fix that, download the repo as a zip and install it with Arduino IDE.
> This will fix your issue.

@fpistm Many thanks - all fixed!!

fpistm, 2019-11-08:
Welcome. My fault

closed 2019-11-08 by fpistm