stm32duino/STM32Ethernet #34: Generic PHY support

fpistm, 2019-11-10:
Currently, LAN8742A_PHY is only supported.
Goal is to have a generic PHY support to be able to support some other.

### requested PHY support:

* [ ] DP83848_PHY #15 
* [ ] LAN8720 PHY #23

sstaub, 2024-02-21:
Is there any progress? The thread started 2019 and meanwhile it is very annoying that nothing happens, especially because most of the supported boards are deprecated.

fpistm, 2024-02-22:
Not at this time. If I can have some bandwidth I'd like to implement it. Do not forget, it is a community project, any contribution are also welcome.

realitytech, 2024-02-24:
In the same boat here. Have found a really handy little board that would be perfect for so many projects but as it uses the DP83848 PHY I'm unable to use it. I'm happy to provide a board if it helps at all.

wowtah, 2024-03-03:
I'm trying to get the library working with a KSZ8863 PHY, but so far haven't been successful yet. So +1 for this request.