stm32duino/STM32Ethernet #36: STM32Ethernet + STM32FreeRTOS interrupt problems

ramboerik, 2020-01-09:
Hi,

My project needs non-blocking Ethernet so I started using this repo with [FreeRTOS](https://github.com/stm32duino/STM32FreeRTOS) to achieve that. I soon encountered problems with the interrupt design of the Ethernet implementation. The board always hang in places where the Ethernet implementation uses `HAL_GetTick` or `HAL_Delay` from its interrupt. I tracked it down and found that the FreeRTOS system interrupt stopped to get called and FreeRTOS internal ticks was never increased, causing the delays to never end.
I tried to fix the issue by increasing the priority of FreeRTOS's interrupt by never got it to work.

What I found to be a working solution is to skip using the Ethernet interrupt and set up a FreeRTOS task to handle calls to `Ethernet.schedule()`. Could you add a define to toggle the interrupt behavior on/off for use with FreeRTOS? or is there any better way to handle this?

fpistm, 2020-01-09:
I guess you talk about this line:
https://github.com/stm32duino/STM32Ethernet/blob/87a5de5262a77146c3d35908d4ff03af8877859a/src/utility/stm32_eth.cpp#L276

You're right this library is not really RTOS compatible.
I think a better way would be to avoid using `HAL_GetTick()` and/or maybe add some RTOS implementation to avoid those issues.

ramboerik, 2020-01-09:
I will probably hit them all sooner or later but the one i got stuck on now is ``HAL_Delay`` in:
https://github.com/stm32duino/Arduino_Core_STM32/blob/ea8664573c69acf7673103b6ef198ef4b22d00b9/system/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_eth.c#L1684

Another thing that didn't work by unknown reason is the `rand()` call in:
https://github.com/stm32duino/LwIP/blob/484e6bbe4f74c6b8f4d73361963b0f7e0d29fb22/src/core/ipv4/dhcp.c#L1913  
I had to compile with `-DDHCP_CREATE_RAND_XID=0` to get a DHCP lease without deadlock.

> I think a better way would be to avoid using HAL_GetTick() and/or maybe add some RTOS  implementation to avoid those issues.  

Okay, I will continue to use my implementation for now but keep that in mind if I get more problems.

Ddh012345dhD, 2021-04-12:
STM32Ethernet and STM32FreeRTOS don't work together in 1 project? How to use them in 1 project?

closed 2022-02-28 by fpistm

tpcorrea, 2023-08-17:
Hi, everyone. I am running at the same issue in my code here. This issue was set as "Done", but I guess the problem wasn´t fixed, right?

This is my set-up:
self-developed board with STM32F407
Windows 11
VS Code, PlatformIO 
stm32duino/STM32duino FreeRTOS@^10.3.2
stm32duino/STM32Ethernet@^1.3.0
stm32duino/STM32duino LwIP@^2.1.2

Is @ramboerik work around the best solution this far?

Thanks a lot and kind regards,
Tomas

fpistm, 2023-08-28:
> Hi, everyone. I am running into the same issue in my code here. This issue was set as "Done", but I guess the problem wasn´t fixed, right?
> 
> This is my set-up: self-developed board with STM32F407 Windows 11 VS Code, PlatformIO stm32duino/STM32duino FreeRTOS@^10.3.2 stm32duino/STM32Ethernet@^1.3.0 stm32duino/STM32duino LwIP@^2.1.2
> 
> Is @ramboerik workaround the best solution this far?
> 
> Thanks a lot and kind regards, Tomas

Tested with Arduino IDE and it works. We do not support PIO. Probably a misalignment of the version.

tpcorrea, 2023-08-28:
@fpistm, would you be so kind and tell me which LWIP, STM32Ethernet and STM32duino versions you get it working?

Thanks a lot and BR,
Tomas

fpistm, 2023-08-28:
I'm always up to date. So last one for all libraries and core.