stm32duino/STM32Ethernet #37: EthernetClient::write cannot send large packages

ramboerik, 2020-01-10:
Hi again 😃 

I'm trying to send a bit larger packages and discovered that it isn't possible with the current implementation. The EthernetClient implementation passes all calls through to LwIP but LwIP's ``tcp_write`` refuses larger packages than ``TCP_SND_BUF = 5840``. The doc for ``tcp_write`` states:

````c
 * The tcp_write() function will fail and return ERR_MEM if the length
 * of the data exceeds the current send buffer size or if the length of
 * the queue of outgoing segment is larger than the upper limit defined
 * in lwipopts.h. The number of bytes available in the output queue can
 * be retrieved with the tcp_sndbuf() function.
 *
 * The proper way to use this function is to call the function with at
 * most tcp_sndbuf() bytes of data. If the function returns ERR_MEM,
 * the application should wait until some of the currently enqueued
 * data has been successfully received by the other host and try again.
 *
````
It should be up to the caller (in this case ``EthernetClient::write``) to make multiple calls to ``tcp_write`` with the data length less than ``tcp_sndbuf()``.

Current ``tcp_write`` call in ``EthernetClient::write``
````c++
99  if(ERR_OK != tcp_write(_tcp_client->pcb, buf, size, TCP_WRITE_FLAG_COPY)) {
100    return 0;
101  }
````

I guess a simple fix for this is to loop over ``tcp_write`` and split the buffer in smaller sizes if data length is larger than ``tcp_sndbuf()``

fpistm, 2020-01-10:
Hi  @ramboerik 
Do not hesitate to provide a PR 😉 
This is a community project so any help are welcome.

ramboerik, 2020-01-10:
Fix available in https://github.com/stm32duino/STM32Ethernet/pull/38. tested with larger packet sizes and it works. However, there is no timeout condition in the write loop and it expects that all data can sooner or later be sent by the stack below. if ``tcp_sndbuf(_tcp_client->pcb)`` would be set to zero for eternity the write loop would hang. I don't know if that can happen but just pointing it out here.

PR #38 (2020-02-21): Fix issue with EthernetClient::write and large packages, closes #37
Fixes #37
commit ae20bcb3 (2020-02-21): Fix issue with EthernetClient::write and large packages, closes #37
commit dc011314 (2020-02-21): Fix issue with EthernetClient::write and large packages

Fixes #37
closed 2020-02-21 by fpistm