stm32duino/STM32Ethernet #42: Ethernet does not work if uC starts with the cable disconnected

rtek1000, 2020-03-06:
Hello, i would like to report a problem, i got around, but it may be necessary to find another better alternative.

https://community.st.com/s/question/0D50X0000CDolzDSQR/ethernet-does-not-work-if-uc-starts-with-the-cable-disconnected

fpistm, 2020-03-09:
Hi @rtek1000 
thanks for report.
We will check this.

ABOSTM, 2020-03-11:
Hi @rtek1000,
I tested on my side and it works like a charm.
Board: NUCLEO_F429ZI which embed PHY: LAN8742
Sketch: WebServer.ino   (provided with this library)

**Test:** 
Supply board while Ethernet cable is unplugged.
Wait a little bit, so that everything setup.
Plug Ethernet cable.
Ping the board with another device.
Response to ping is correctly received by the other device.

rtek1000, 2020-03-11:
Please test again, the wait time I'm using is 10 seconds or more. The LwIP startup has a long timeout, 5 seconds if I'm not mistaken.

ABOSTM, 2020-03-13:
Hi @rtek1000,
I reproduce your issue, but it was not so easy:
As you said I need to wait around 10 seconds or more.
But even if I wait, it is not systematic.
Also, most of the time, when debugger is attached, I could not reproduce.
And finally it seems that depending on the debug code I added, the problem was more or less reproducible.

I pushed a PR #44 to rework the management of ethernet cable hotplug, and make it similar to NUCLEO-H743ZI Cube Application: LwIP_HTTP_Server_Netconn_RTOS

I tested this patch on my setup NUCLEO_F429ZI which embed LAN8742, but as far as I can see it should be compatible with your LAN8720.
Can you test it and tell me if this solve your issue ?

rtek1000, 2020-03-15:
Hi,

I looked at the file but it seems to already have these settings, do I have to make any changes?

[image]

rtek1000, 2020-03-15:
I would like to inform you that I tested ESP32 with LAN8720 and this disconnected cable problem does not occur with ESP32.

Code used to test:
https://github.com/espressif/arduino-esp32/issues/3819

ABOSTM, 2020-03-16:
Hi @rtek1000 ,
Yes I would like you to test the PR #44
Modification have nothing to do with `#define IFNAME0 ...`
instead, it  concerns 2 files with tens of modifications.

rtek1000, 2020-03-16:
Hi,

I downloaded the STM32Ethernet package again and replaced it in the libraries folder,

After compiling and uploading, these were the results:

Connected cable at start:

> 08:10:56.131 -> Begin Ethernet
> 08:11:00.303 -> Success to configure Ethernet using DHCP

Disconnected cable at start:

> 08:11:00.303 -> localIP: 192.168.1.103
> 08:11:00.303 -> subnetMask: 255.255.255.0
> 08:11:00.303 -> gatewayIP: 192.168.1.1
> 08:11:00.303 -> dnsServerIP: 192.168.1.1
> 08:12:04.793 -> Begin Ethernet
> 08:13:10.369 -> Failed to configure Ethernet using DHCP

I left the window dripping on the IP found the first time, but received no response.

To prevent the DHCP service from providing a different address, I configured my router to always provide the same IP address.

[image]

Connected cable at start:
> 08:21:26.352 -> Begin Ethernet
> 08:21:29.445 -> Success to configure Ethernet using DHCP
> 08:21:29.445 -> localIP: 192.168.1.149
> 08:21:29.445 -> subnetMask: 255.255.255.0
> 08:21:29.445 -> gatewayIP: 192.168.1.1
> 08:21:29.445 -> dnsServerIP: 192.168.1.1
> 08:23:21.275 -> Seconds since Jan 1 1900 = 3793346601
> 08:23:21.275 -> Unix time = 1584357801
> 08:23:32.338 -> The UTC time is 11:23:32

Disconnected cable at start:
> 08:24:57.309 -> Begin Ethernet
> 08:26:02.921 -> Failed to configure Ethernet using DHCP

It seems that this code still needs to perform the reset after connecting the cable, otherwise, it is not possible to use the ethernet network.

Sketch:
```C++
/*

 Udp NTP Client

 Get the time from a Network Time Protocol (NTP) time server
 Demonstrates use of UDP sendPacket and ReceivePacket
 For more on NTP time servers and the messages needed to communicate with them,
 see http://en.wikipedia.org/wiki/Network_Time_Protocol

 created 4 Sep 2010
 by Michael Margolis
 modified 9 Apr 2012
 by Tom Igoe
 modified 02 Sept 2015
 by Arturo Guadalupi
 modified 23 Jun 2017
 by Wi6Labs
 modified 1 Jun 2018
 by sstaub
 This code is in the public domain.

 */

#include <LwIP.h>
#include <STM32Ethernet.h>
[... 288 lines cut ...]
}

// send an NTP request to the time server at the given address
void sendNTPpacket(char* address) {
  // set all bytes in the buffer to 0
  memset(packetBuffer, 0, NTP_PACKET_SIZE);
  // Initialize values needed to form NTP request
  // (see URL above for details on the packets)
  packetBuffer[0] = 0b11100011;   // LI, Version, Mode
  packetBuffer[1] = 0;     // Stratum, or type of clock
  packetBuffer[2] = 6;     // Polling Interval
  packetBuffer[3] = 0xEC;  // Peer Clock Precision
  // 8 bytes of zero for Root Delay & Root Dispersion
  packetBuffer[12]  = 49;
  packetBuffer[13]  = 0x4E;
  packetBuffer[14]  = 49;
  packetBuffer[15]  = 52;

  // all NTP fields have been given values, now
  // you can send a packet requesting a timestamp:
  Udp.beginPacket(address, 123); //NTP requests are to port 123
  Udp.write(packetBuffer, NTP_PACKET_SIZE);
  Udp.endPacket();
}
```

ABOSTM, 2020-03-16:
> I downloaded the STM32Ethernet package again and replaced it in the libraries folder,

I am not confident about the way you get the library. Can you verify that in your source file src/utility/ethernetif.cpp there is the following define:
`#define PHY_SR_AUTODONE ((uint16_t)0x1000)`

If it is not the case, you can get the modified files version:
* https://github.com/stm32duino/STM32Ethernet/blob/dd7d8acbe5a57279746d5a5c9ca7f284782ad84b/src/utility/ethernetif.cpp
* https://github.com/stm32duino/STM32Ethernet/blob/dd7d8acbe5a57279746d5a5c9ca7f284782ad84b/src/utility/stm32_eth.cpp

rtek1000, 2020-03-16:
It worked,

But I think I made a mistake, sorry, I've been comparing incompatible codes. The STM32 code uses static IP and the Arduino code uses DHCP.

I added a static IP to the previous sketch, and now I can say that the ping is getting a response after connecting the cable, if the cable was disconnected when starting the microcontroller.

I tested the Arduino library with the modification (replacing the contents of the two files) and also tested the library again without the modification. Both worked.

For the Arduino case, it must have been the lack of obtaining the IP address via DHCP. After I added another attempt to initialize Ethernet, both codes worked, using DHCP and with static IP.

Unfortunately, unlike Arduino, the STM32CubeMX (or IDE) code still doesn't work, for static IP. I will inform them that the Arduino code works, and that it can serve as a basis for locating the problem.

Thank you.

```C++
/*

  Udp NTP Client

  Get the time from a Network Time Protocol (NTP) time server
  Demonstrates use of UDP sendPacket and ReceivePacket
  For more on NTP time servers and the messages needed to communicate with them,
  see http://en.wikipedia.org/wiki/Network_Time_Protocol

  created 4 Sep 2010
  by Michael Margolis
  modified 9 Apr 2012
  by Tom Igoe
  modified 02 Sept 2015
  by Arturo Guadalupi
  modified 23 Jun 2017
  by Wi6Labs
  modified 1 Jun 2018
  by sstaub
  modified 16 Mar 2020
  by rtek1000
  This code is in the public domain.

*/

#include <LwIP.h>
#include <STM32Ethernet.h>
#include <EthernetUdp.h>

//#define USE_DHCP 1

#ifdef USE_DHCP
bool Ethernet_IsInitialized = false;
#endif

#ifndef USE_DHCP
// Enter an IP address for your controller below.
// The IP address will be dependent on your local network.
// gateway and subnet are optional:

IPAddress ip(192, 168, 1, 177);
IPAddress myDns(192, 168, 1, 1);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
#endif

unsigned int localPort = 8888;       // local port to listen for UDP packets

char timeServer[] = "time.nist.gov"; // time.nist.gov NTP server

const int NTP_PACKET_SIZE = 48; // NTP time stamp is in the first 48 bytes of the message

byte packetBuffer[ NTP_PACKET_SIZE]; //buffer to hold incoming and outgoing packets

// A UDP instance to let us send and receive packets over UDP
[... 113 lines cut ...]
#endif

// send an NTP request to the time server at the given address
void sendNTPpacket(char* address) {
  // set all bytes in the buffer to 0
  memset(packetBuffer, 0, NTP_PACKET_SIZE);
  // Initialize values needed to form NTP request
  // (see URL above for details on the packets)
  packetBuffer[0] = 0b11100011;   // LI, Version, Mode
  packetBuffer[1] = 0;     // Stratum, or type of clock
  packetBuffer[2] = 6;     // Polling Interval
  packetBuffer[3] = 0xEC;  // Peer Clock Precision
  // 8 bytes of zero for Root Delay & Root Dispersion
  packetBuffer[12]  = 49;
  packetBuffer[13]  = 0x4E;
  packetBuffer[14]  = 49;
  packetBuffer[15]  = 52;

  // all NTP fields have been given values, now
  // you can send a packet requesting a timestamp:
  Udp.beginPacket(address, 123); //NTP requests are to port 123
  Udp.write(packetBuffer, NTP_PACKET_SIZE);
  Udp.endPacket();
}
```

PR #44 (2020-03-19): Rework ethernet cable hotplug detection
Rework ethernet cable hotplug detection

Mimic NUCLEO-H743ZI Cube Application: LwIP_HTTP_Server_Netconn_RTOS
Fixes #42
commit 6dd579b3 (2020-03-19): Rework ethernet cable hotplug detection

Mimic NUCLEO-H743ZI Cube Application: LwIP_HTTP_Server_Netconn_RTOS

Fixes #42
closed 2020-03-16 by rtek1000