stm32duino/STM32Ethernet #45: stm32_eth_scheduler() and timer interrupt

manitou48, 2020-05-08:
I'm looking at porting your arduino-like lib and your LwIP lib for use with Teensy 3.5/3.6/4.1.  I have an lwip implmentation that uses callbacks and polling with the Teensy.   I haven't run any tests with your libs yet, so my question is based on code inspection.  As I understand your non-RTOS implementation, there seems to be a potential conflict (race condition) with the timer-based (1 ms) calls to stm32_eth_scheduler() and the calls to the scheduler embedded in DHCP and UDP modules. It would seem a timer-based  call to the scheduler could interrupt a scheduler invocation from DHCP or UDP.  Thanks for any enlightenment.

fpistm, 2020-05-11:
Hi @manitou48 
yes this is not RTOS oriented.
Which _scheduler invocation from DHCP or UDP_ you talk about ?

manitou48, 2020-05-11:
here is list of calls to stm32_eth_scheduler (inclusing the TIMER based call)
```
  stm32_eth_scheduler();
  stm32_eth_scheduler();
  stm32_eth_scheduler();
  stm32_eth_scheduler();
  stm32_eth_scheduler();
./EthernetUdp.cpp
  stm32_eth_scheduler();
    // stm32_eth_scheduler() will be called every 1ms.
  stm32_eth_scheduler();
void stm32_eth_scheduler(void)
        stm32_eth_scheduler();
./utility/stm32_eth.cpp
  stm32_eth_scheduler();
    stm32_eth_scheduler();
  stm32_eth_scheduler();
./Dhcp.cpp
    stm32_eth_scheduler();
    stm32_eth_scheduler();
  stm32_eth_scheduler();
  stm32_eth_scheduler();
./EthernetClient.cpp
  stm32_eth_scheduler();
./STM32Ethernet.cpp
```
there is also  Ethernet.schedule(), though none of the examples use it.

maybe stm32_eth_scheduler() is re-entrant?

fpistm, 2020-05-11:
Ah ok.
So yes a call to `stm32_eth_scheduler` can be interrupted by the timer IRQ call.
I could not tell if this is an issue, this will need more investigations anyway no related issue was raised around this. This is like this since the library was created but as said this should requireed more investigations to tell if this could be an issue mainly when `ETH_INPUT_USE_IT` and `LWIP_DHCP` are enabled.
Feel free to contribut on this 😉

raphaelvalentin, 2021-08-05:
I am experiencing a random firmware hard crash of the F767 MCU and this `race condition` could be "potentially" the reason. 
In my test case, I have a static page `css + HTML + 6 images` for a total document size of ~350kB (data are read from an external serial nor flash via a custom `FS`). This document is about documentation and the final HTML will use the HTML lazy loading feature.
I reduced the value of the timer from 1ms to 50us (increasing the number of events) and plugged many `Serial.print` to narrow the location of the crash.
Also, I called the function scheduler in my main loop sketch such if a delay is over 50us, scheduler is called. This has the effect to randomize a call of the scheduler from the timer and cancel a predictable delay call.
Packets of 512 bytes are read from the `FS` and written to `EthernetClient::write`.
Typically, `ping` is reduced to 150us and the connection speed reaches a value of 1MB/s on average (`wget`).
I finally narrowed down the crash issue location in the function `EthernetClient::write`.
(1) When I turned on the Timer (default), the crash is happening after 5-10 document loading processes from the browser (cache disabled; nothing regular at first view).
(2) When I turned off the Timer (comment the line `// TIM_scheduler_Config();`), no more crashes.
Unfortunately, I experienced issues getting an exact location and I am not sure if it comes from `tcp_sndbuf`, `tcp_write`, `tcp_output`, or `stm32_eth_scheduler`. Usually, it crashes between `stm32_eth_scheduler` and `tcp_sndbuf`.
Maybe, you may have a point of view about my problem.
Thanks a lot for your support, 
Raphael.

closed 2022-03-14 by fpistm

fpistm, 2022-03-14:
Hi @manitou48,

@ABOSTM made a PR to avoid this. Now each call of `stm32_eth_scheduler` trig the interrupt handler.

commit e2e0a42d (2022-03-14): fix: call effective ETH scheduler only from Timer callback

Restrict effective ETH scheduler to Timer callback only.
This is to avoid any race condition on ETH scheduler,
which could previously be called from timer ISR,
as well as  DHCP or UDP driver parts.
Any direct request for ETH scheduler (direct call to stm32_eth_scheduler())
will generate a Timer Update Event to force a call to timer callback

Fixes #45