stm32duino/STM32Ethernet #46: Compile fails with version 1.9.0 of STM32 Core

prototypicalpro, 2020-05-21:
STM32Ethernet fails to compile with version `v1.9.0` of the [Arduino STM32 core](https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/1.9.0) with the following error:
```
/home/travis/Arduino/libraries/STM32duino_STM32Ethernet/src/utility/stm32_eth.cpp: In function 'void TIM_scheduler_Config()':
/home/travis/Arduino/libraries/STM32duino_STM32Ethernet/src/utility/stm32_eth.cpp:179:45: error: no matching function for call to 'HardwareTimer::attachInterrupt(void (&)(HardwareTimer*))'
  179 |   EthTim->attachInterrupt(scheduler_callback);
```

This seems to be because the STM32 core changed the parameter type of `void HardwareTimer::attachInterrupt` from a function pointer to a function object in the `v1.9.0` update: https://github.com/stm32duino/Arduino_Core_STM32/commit/2983175adc4162efcc88d7ed22cf310d21edb002. `v1.8.0` of the STM32 core does not have this change and compiles fine.

fpistm, 2020-05-21:
Right, I forgot to do a new release. I will do soon.

fpistm, 2020-05-22:
Release 1.2.0 done:
https://github.com/stm32duino/STM32Ethernet/releases/tag/1.2.0

closed 2020-05-22 by fpistm