stm32duino/STM32Ethernet #55: How to fix?

Ddh012345dhD, 2021-04-02:
[image]

fpistm, 2021-04-02:
Hi @Ddh012345dhD 
we do not support PIO here only Arduino IDE.
Try with Arduino IDE if it is the same then we could investigate. Anyway I know it is ok.

Ddh012345dhD, 2021-04-02:
I try Adruino IDE.
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:80:15: error: 'ETH_DMADescTypeDef' does not name a type
   80 | __ALIGN_BEGIN ETH_DMADescTypeDef  DMARxDscrTab[ETH_RXBUFNB] __ALIGN_END;/* Ethernet Rx MA Descriptor */
      |               ^~~~~~~~~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:85:15: error: 'ETH_DMADescTypeDef' does not name a type
   85 | __ALIGN_BEGIN ETH_DMADescTypeDef  DMATxDscrTab[ETH_TXBUFNB] __ALIGN_END;/* Ethernet Tx DMA Descriptor */
      |               ^~~~~~~~~~~~~~~~~~
In file included from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/STM32F4xx/stm32f4xx_hal_conf.h:13,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:30,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h:250,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/stm32_def.h:34,
                 from C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:48:
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/STM32F4xx/stm32f4xx_hal_conf_default.h:227:40: error: 'ETH_MAX_PACKET_SIZE' was not declared in this scope
  227 | #define ETH_RX_BUF_SIZE                ETH_MAX_PACKET_SIZE /* buffer size for receive               */
      |                                        ^~~~~~~~~~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:90:44: note: in expansion of macro 'ETH_RX_BUF_SIZE'
   90 | __ALIGN_BEGIN uint8_t Rx_Buff[ETH_RXBUFNB][ETH_RX_BUF_SIZE] __ALIGN_END; /* Ethernet Receive Buffer */
      |                                            ^~~~~~~~~~~~~~~
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/STM32F4xx/stm32f4xx_hal_conf_default.h:228:40: error: 'ETH_MAX_PACKET_SIZE' was not declared in this scope
  228 | #define ETH_TX_BUF_SIZE                ETH_MAX_PACKET_SIZE /* buffer size for transmit              */
      |                                        ^~~~~~~~~~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:95:44: note: in expansion of macro 'ETH_TX_BUF_SIZE'
   95 | __ALIGN_BEGIN uint8_t Tx_Buff[ETH_TXBUFNB][ETH_TX_BUF_SIZE] __ALIGN_END; /* Ethernet Transmit Buffer */
      |                                            ^~~~~~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:97:8: error: 'ETH_HandleTypeDef' does not name a type; did you mean 'RTC_HandleTypeDef'?
   97 | static ETH_HandleTypeDef EthHandle;
      |        ^~~~~~~~~~~~~~~~~
      |        RTC_HandleTypeDef
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:116:22: error: variable or field 'HAL_ETH_MspInit' declared void
  116 | void HAL_ETH_MspInit(ETH_HandleTypeDef *heth)
      |                      ^~~~~~~~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:116:22: error: 'ETH_HandleTypeDef' was not declared in this scope; did you mean 'RTC_HandleTypeDef'?
  116 | void HAL_ETH_MspInit(ETH_HandleTypeDef *heth)
      |                      ^~~~~~~~~~~~~~~~~
      |                      RTC_HandleTypeDef
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:116:41: error: 'heth' was not declared in this scope
  116 | void HAL_ETH_MspInit(ETH_HandleTypeDef *heth)
      |                                         ^~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp: In function 'void low_level_init(netif*)':
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:168:3: error: 'EthHandle' was not declared in this scope
  168 |   EthHandle.Instance = ETH;
      |   ^~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:170:36: error: 'ETH_AUTONEGOTIATION_ENABLE' was not declared in this scope
  170 |   EthHandle.Init.AutoNegotiation = ETH_AUTONEGOTIATION_ENABLE;
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:171:26: error: 'ETH_SPEED_100M' was not declared in this scope
  171 |   EthHandle.Init.Speed = ETH_SPEED_100M;
      |                          ^~~~~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:172:31: error: 'ETH_MODE_FULLDUPLEX' was not declared in this scope
  172 |   EthHandle.Init.DuplexMode = ETH_MODE_FULLDUPLEX;
      |                               ^~~~~~~~~~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:174:35: error: 'ETH_MEDIA_INTERFACE_RMII' was not declared in this scope
  174 |   EthHandle.Init.MediaInterface = ETH_MEDIA_INTERFACE_RMII;
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master\src\utility\ethernetif.cpp:181:27: error: 'ETH_RXPOLLING_MODE' was not declared in this scope
  181 |   EthHandle.Init.RxMode = ETH_RXPOLLING_MODE;
[... cut ...]
  584 |     HAL_ETH_Stop(&EthHandle);
      |     ^~~~~~~~~~~~
      |     HAL_DAC_Stop
Using library LwIP-master at version 2.1.2 in folder: C:\Users\PC\Documents\Arduino\libraries\LwIP-master 
Using library STM32Ethernet-master at version 1.2.0 in folder: C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master 
Using library SrcWrapper at version 1.0.1 in folder: C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper 
exit status 1
Error compiling for board Generic STM32F4 series.

fpistm, 2021-04-02:
OK. The generic F407 does not enable the `HAL_ETH_MODULE_ENABLED`

https://github.com/stm32duino/wiki/wiki/HAL-configuration#customize-hal-or-variant-definition

So you have to enable it. Probably you also need to redefined the `PinMap_Ethernet` array in order to have only the requested pin.

https://github.com/stm32duino/wiki/wiki/Custom-definitions#custom-pinmap-array

I've tested and it build OK.

closed 2021-04-02 by fpistm

Ddh012345dhD, 2021-04-02:
In file included from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h:250,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/stm32_def.h:34,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/clock.h:43,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring_time.h:23,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring.h:38,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/Arduino.h:36,
                 from C:\Users\PC\AppData\Local\Temp\arduino_build_997497\sketch\DhcpAddressPrinter.ino.cpp:1:
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:220:1: error: 'HAL_StatusTypeDef' does not name a type
  220 | HAL_StatusTypeDef HAL_Init(void);
      | ^~~~~~~~~~~~~~~~~
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:221:1: error: 'HAL_StatusTypeDef' does not name a type
  221 | HAL_StatusTypeDef HAL_DeInit(void);
      | ^~~~~~~~~~~~~~~~~
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:224:1: error: 'HAL_StatusTypeDef' does not name a type
  224 | HAL_StatusTypeDef HAL_InitTick (uint32_t TickPriority);
      | ^~~~~~~~~~~~~~~~~
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:237:1: error: 'HAL_StatusTypeDef' does not name a type
  237 | HAL_StatusTypeDef HAL_SetTickFreq(HAL_TickFreqTypeDef Freq);
      | ^~~~~~~~~~~~~~~~~
In file included from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:31,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/analog.h:45,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/board.h:8,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring.h:41,
                 from C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/Arduino.h:36,
                 from C:\Users\PC\AppData\Local\Temp\arduino_build_997497\sketch\DhcpAddressPrinter.ino.cpp:1:
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/timer.h:245:3: error: 'TIM_HandleTypeDef' does not name a type; did you mean 'TIM_TypeDef'?
  245 |   TIM_HandleTypeDef handle;
      |   ^~~~~~~~~~~~~~~~~
      |   TIM_TypeDef
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/timer.h:251:27: error: 'TIM_HandleTypeDef' was not declared in this scope; did you mean 'TIM_TypeDef'?
  251 | timerObj_t *get_timer_obj(TIM_HandleTypeDef *htim);
      |                           ^~~~~~~~~~~~~~~~~
      |                           TIM_TypeDef
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/timer.h:251:46: error: 'htim' was not declared in this scope; did you mean 'tm'?
  251 | timerObj_t *get_timer_obj(TIM_HandleTypeDef *htim);
      |                                              ^~~~
      |                                              tm
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/timer.h:253:23: error: variable or field 'enableTimerClock' declared void
  253 | void enableTimerClock(TIM_HandleTypeDef *htim);
      |                       ^~~~~~~~~~~~~~~~~
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/timer.h:253:23: error: 'TIM_HandleTypeDef' was not declared in this scope; did you mean 'TIM_TypeDef'?
  253 | void enableTimerClock(TIM_HandleTypeDef *htim);
      |                       ^~~~~~~~~~~~~~~~~
      |                       TIM_TypeDef
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/timer.h:253:42: error: 'htim' was not declared in this scope; did you mean 'tm'?
  253 | void enableTimerClock(TIM_HandleTypeDef *htim);
      |                                          ^~~~
      |                                          tm
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/timer.h:254:24: error: variable or field 'disableTimerClock' declared void
  254 | void disableTimerClock(TIM_HandleTypeDef *htim);
      |                        ^~~~~~~~~~~~~~~~~
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/timer.h:254:24: error: 'TIM_HandleTypeDef' was not declared in this scope; did you mean 'TIM_TypeDef'?
  254 | void disableTimerClock(TIM_HandleTypeDef *htim);
      |                        ^~~~~~~~~~~~~~~~~
      |                        TIM_TypeDef
C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/timer.h:254:43: error: 'htim' was not declared in this scope; did you mean 'tm'?
  254 | void disableTimerClock(TIM_HandleTypeDef *htim);
      |                                           ^~~~
[... cut ...]
   69 |   UART_HandleTypeDef handle;
      |   ^~~~~~~~~~~~~~~~~~
Using library LwIP-master at version 2.1.2 in folder: C:\Users\PC\Documents\Arduino\libraries\LwIP-master 
Using library STM32Ethernet-master at version 1.2.0 in folder: C:\Users\PC\Documents\Arduino\libraries\STM32Ethernet-master 
Using library SrcWrapper at version 1.0.1 in folder: C:\Users\PC\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\libraries\SrcWrapper 
exit status 1
Error compiling for board Generic STM32F4 series.

fpistm, 2021-04-02:
Don't know what you made but you have to simply add a file name `hal_conf_extra.h` with inside:
`#define HAL_ETH_MODULE_ENABLED`

Then you will be able to compile.
Here the result of the webclient example built for Generic F407 

```
Using library STM32duino_LwIP at version 2.1.2 in folder: C:\STM32\arduino\arduino-1.8.13\portable\sketchbook\libraries\STM32duino_LwIP 
Using library STM32duino_STM32Ethernet at version 1.2.0 in folder: C:\STM32\arduino\arduino-1.8.13\portable\sketchbook\libraries\STM32duino_STM32Ethernet 
Using library SrcWrapper at version 1.0.1 in folder: C:\STM32\arduino\arduino-1.8.13\portable\packages\STM32\hardware\stm32\2.0.0-dev\libraries\SrcWrapper 

Sketch uses 55256 bytes (5%) of program storage space. Maximum is 1048576 bytes.
Global variables use 44168 bytes (33%) of dynamic memory, leaving 86904 bytes for local variables. Maximum is 131072 bytes.
```