stm32duino/STM32Ethernet #56: Compilation error on freshly installed Arduino 1.8.16 with Nucleo-144 Nucleo F207ZG

N0rbert, 2021-11-06:
I have just installed Arduino 1.8.16 on fresh Ubuntu 18.04 LTS VM.
Then I have installed "STM32 MCU based boards" 2.1.0 BSP according to https://github.com/stm32duino/wiki/wiki/Getting-Started .
I visited libraries and insalled STM32duino STM32Ethernet library at version 1.2.0 (with LwIP 2.1.2 as dependency).
Then I have selected my STM32 Nulceo-144 Nucleo F207ZG as target board.
Opened DHCPAddressPrinter example to compile it. Got the following errors:

```
/home/mate/Arduino/libraries/STM32duino_STM32Ethernet/src/utility/ethernetif.cpp: In function 'void low_level_init(netif*)':
/home/mate/Arduino/libraries/STM32duino_STM32Ethernet/src/utility/ethernetif.cpp:184:31: error: 'LAN8742A_PHY_ADDRESS' was not declared in this scope
  184 |   EthHandle.Init.PhyAddress = LAN8742A_PHY_ADDRESS;
      |                               ^~~~~~~~~~~~~~~~~~~~
/home/mate/Arduino/libraries/STM32duino_STM32Ethernet/src/utility/ethernetif.cpp:223:39: error: 'PHY_IMR' was not declared in this scope; did you mean 'PHY_MISR'?
  223 |   HAL_ETH_ReadPHYRegister(&EthHandle, PHY_IMR, &regvalue);
      |                                       ^~~~~~~
      |                                       PHY_MISR
/home/mate/Arduino/libraries/STM32duino_STM32Ethernet/src/utility/ethernetif.cpp:225:15: error: 'PHY_ISFR_INT4' was not declared in this scope
  225 |   regvalue |= PHY_ISFR_INT4;
      |               ^~~~~~~~~~~~~
/home/mate/Arduino/libraries/STM32duino_STM32Ethernet/src/utility/ethernetif.cpp: In function 'void ethernetif_set_link(netif*)':
/home/mate/Arduino/libraries/STM32duino_STM32Ethernet/src/utility/ethernetif.cpp:509:39: error: 'PHY_ISFR' was not declared in this scope; did you mean 'PHY_MISR'?
  509 |   HAL_ETH_ReadPHYRegister(&EthHandle, PHY_ISFR, &regvalue);
      |                                       ^~~~~~~~
      |                                       PHY_MISR
/home/mate/Arduino/libraries/STM32duino_STM32Ethernet/src/utility/ethernetif.cpp:512:19: error: 'PHY_ISFR_INT4' was not declared in this scope
  512 |   if ((regvalue & PHY_ISFR_INT4) != (uint16_t)RESET) {
      |                   ^~~~~~~~~~~~~
exit status 1
Error compiling for board Nucleo-144.
```

What am I doing wrong here?

N0rbert, 2021-11-06:
Full [error-log.txt](https://github.com/stm32duino/STM32Ethernet/files/7490362/error-log.txt) if needed.

N0rbert, 2021-11-07:
Really the situation is even worse, DhcpAddressPrinter compiles normally only of 4 all Nucleo-144 famly:

Board | Status
-- | --
Nucleo 			F207ZG | fail
Nucleo 			F429ZI | ok
Nucleo 			F746ZG | ok
Nucleo 			F756ZG | ok
Nucleo 			F767ZI | ok
Nucleo 			H743ZI | fail
Nucleo 			H743ZI2 | fail
Nucleo 			L496ZG | fail
Nucleo 			L496ZG-P | fail
Nucleo 			L4R5ZI | fail
Nucleo 			L4R5ZI-P | fail
Nucleo 			L552ZE-Q | fail
Nucleo 			U575ZI-Q | fail

fpistm, 2021-11-10:
Hi @N0rbert,

In fact issue comes from the PHY definition avaialble in the default hal conf file for F2, it could be simply added as it is supported: LAN8742A
For other target the PHY is different and also the HAL ethernet has been reworked in the cube and need some rework here:
https://github.com/stm32duino/STM32Ethernet/issues/34
https://github.com/stm32duino/STM32Ethernet/issues/21

I've made a PR in the core to update the hal configuration:
https://github.com/stm32duino/Arduino_Core_STM32/pull/1540

fpistm, 2021-11-10:
I close this issue as no update will be done in this library and it build properly with the correct configuration:

```console
Using library STM32duino_LwIP at version 2.1.2 in folder: C:\STM32\arduino\arduino-1.8.16\portable\sketchbook\libraries\STM32duino_LwIP 
Using library STM32duino_STM32Ethernet at version 1.2.0 in folder: C:\STM32\arduino\arduino-1.8.16\portable\sketchbook\libraries\STM32duino_STM32Ethernet 
Using library SrcWrapper at version 1.0.1 in folder: C:\STM32\arduino\arduino-1.8.16\portable\packages\STMicroelectronics\hardware\stm32\2.1.0\libraries\SrcWrapper 
"C:\\STM32\\arduino\\arduino-1.8.16\\portable\\packages\\STMicroelectronics\\tools\\xpack-arm-none-eabi-gcc\\10.2.1-1.1/bin/arm-none-eabi-size" -A "C:\\Users\\AppData\\Local\\Temp\\arduino_build_954025/DhcpAddressPrinter.ino.elf"
Sketch uses 52444 bytes (5%) of program storage space. Maximum is 1048576 bytes.
Global variables use 41020 bytes (31%) of dynamic memory, leaving 90052 bytes for local variables. Maximum is 131072 bytes.

```

closed 2021-11-10 by fpistm

N0rbert, 2021-11-10:
Currently I own only Nucleo-144 F207ZG and H743ZI2 .
F207ZG is now fixed, thank you!