stm32duino/STM32Ethernet #59: Error initializing ethernet client

nicksterious, 2022-03-10:
Not sure if this belongs here but with STM32 core v2.2.0 I am getting this compile error:

`src\utility\stm32_eth.cpp:179:45: error: no matching function for call to 'HardwareTimer::attachInterrupt(void (&)(HardwareTimer*))'`

Full stack trace:

```
H:\My Documents\Arduino\libraries\STM32duino_STM32Ethernet\src\utility\stm32_eth.cpp: In function 'void TIM_scheduler_Config()':
H:\My Documents\Arduino\libraries\STM32duino_STM32Ethernet\src\utility\stm32_eth.cpp:179:45: error: no matching function for call to 'HardwareTimer::attachInterrupt(void (&)(HardwareTimer*))'
  179 |   EthTim->attachInterrupt(scheduler_callback);
      |                                             ^
In file included from H:\My Documents\ArduinoData\packages\STMicroelectronics\hardware\stm32\2.2.0\cores\arduino/stm32/analog.h:45,
                 from H:\My Documents\ArduinoData\packages\STMicroelectronics\hardware\stm32\2.2.0\cores\arduino/board.h:8,
                 from H:\My Documents\ArduinoData\packages\STMicroelectronics\hardware\stm32\2.2.0\cores\arduino/wiring.h:41,
                 from H:\My Documents\ArduinoData\packages\STMicroelectronics\hardware\stm32\2.2.0\cores\arduino/Arduino.h:36,
                 from H:\My Documents\Arduino\libraries\STM32duino_STM32Ethernet\src\utility\stm32_eth.cpp:39:
H:\My Documents\ArduinoData\packages\STMicroelectronics\hardware\stm32\2.2.0\cores\arduino/HardwareTimer.h:137:10: note: candidate: 'void HardwareTimer::attachInterrupt(callback_function_t)'
  137 |     void attachInterrupt(callback_function_t callback); // Attach interrupt callback which will be called upon update event (timer rollover)
      |          ^~~~~~~~~~~~~~~
H:\My Documents\ArduinoData\packages\STMicroelectronics\hardware\stm32\2.2.0\cores\arduino/HardwareTimer.h:137:46: note:   no known conversion for argument 1 from 'void(HardwareTimer*)' to 'callback_function_t' {aka 'std::function<void()>'}
  137 |     void attachInterrupt(callback_function_t callback); // Attach interrupt callback which will be called upon update event (timer rollover)
      |                          ~~~~~~~~~~~~~~~~~~~~^~~~~~~~
H:\My Documents\ArduinoData\packages\STMicroelectronics\hardware\stm32\2.2.0\cores\arduino/HardwareTimer.h:141:10: note: candidate: 'void HardwareTimer::attachInterrupt(uint32_t, callback_function_t)'
  141 |     void attachInterrupt(uint32_t channel, callback_function_t callback); // Attach interrupt callback which will be called upon compare match event of specified channel
      |          ^~~~~~~~~~~~~~~
H:\My Documents\ArduinoData\packages\STMicroelectronics\hardware\stm32\2.2.0\cores\arduino/HardwareTimer.h:141:10: note:   candidate expects 2 arguments, 1 provided
exit status 1
Error compiling for board Generic STM32F4 series.

```

Is there a fix for this?

ABOSTM, 2022-03-10:
Hi @nicksterious
Can you tell me which version of "STM32duino STM32Ethernet" you have installed ?
(menu: Sketch --> Include Library --> Manage Libraries ...)

nicksterious, 2022-03-10:
Hi @ABOSTM 

I was using 1.1.0 which got installed by the library manager, after updating to 1.2.0 the error is gone.

Thank you sir

closed 2022-03-10 by nicksterious