stm32duino/STM32Ethernet #64: no DHCP when STM32Ethernet + STM32Arduino-FreeRTOS?

raphaelvalentin, 2022-08-30:
Hi STM, 
I would like to use DHCP with Ethernet with FreeRTOS for a `STM32F767` MCU. Unfortunatly, my simple example seems not working at this status.

```
//#include <STM32FreeRTOS.h>
#include <LwIP.h>
#include <STM32Ethernet.h>

void setup() {
    Serial.begin(9600);
    delay(1000);
    Serial.println("Start...");
    if (Ethernet.begin() == 0) {
      Serial.println("Failed to configure Ethernet using DHCP");
      while (1);
    }
    Serial.println(Ethernet.localIP());
}
```

From the code below:
(1) When the code does not include `STM32FreeRTOS.h`, the example works well with DHCP; 
(2) However, when the code includes  `STM32FreeRTOS.h`, the example does not work with DHCP but only with a static IP.

This is a strange situation because the 2 libraries shall be uncorrelated (maybe a common variable?). Is there an explanation to this issue and way to fix it?
Thanks a lot for your support !
Raphael.

fpistm, 2022-08-30:
Hi @raphaelvalentin,

Just including STM32FreeRTOS will change some behavior, heap managements, delay implementation so you have to properly configure FreeRTOS and use it (at least start the scheduler.

raphaelvalentin, 2022-08-30:
Understand. Thanks a lot for your fast reply. Actually, I am using your example in your library as a reference. I have a full project that I made working with RTOS and a LAN8720 chip on a custom board working well with static IP. On the side, I have as well the nucleo as reference. I continue my experiments. I will come back.

raphaelvalentin, 2022-08-31:
Hi @fpistm

I understand my issue. The right code shall be like as described below. 
(1) the `Ethernet.begin`  shall be included in a task,
(2) the task shall loop indefinitely and not quit,
(3) it works with both `Generic F767` and `Nucleo 144` variants.
So I think the topic is closed on my issue.

Thanks again for your reply that have put me in right direction.

```
#include <LwIP.h>
#include <STM32Ethernet.h>
#include <STM32FreeRTOS.h>

// default IP
IPAddress ip(192, 168, 1, 177);

void taskETH(void* arg) {
    UNUSED(arg);
    if (Ethernet.begin() == 0) {  // -->> important to place it in a task
        Serial.println("Failed to configure Ethernet using DHCP");
        Ethernet.begin(ip);  // -->> important to place it in a task
    }
    while (1)
        ;  // -->> important to loop to keep the Ethernet connection.
}

void taskPrintIP(void* arg) {
    UNUSED(arg);
    while (Ethernet.localIP() == INADDR_NONE) {
        vTaskDelay(pdMS_TO_TICKS(500));
        Serial.println("Ethernet is not connected");
    }
    while (1) {
        Serial.println(Ethernet.localIP());
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}

void setup() {
    Serial.begin(230400);
    delay(1000);
    Serial.println("Start...");

    xTaskCreate(taskETH, NULL, 200, NULL, 1, NULL);
    xTaskCreate(taskPrintIP, NULL, 200, NULL, 1, NULL);
    vTaskStartScheduler();
    Serial.println("Scheduler failed");
    while (1)
        ;
}

void loop() {}
```

fpistm, 2022-08-31:
Welcome. 
Great to read it works. Thanks for your feedback.

closed 2022-08-31 by fpistm