stm32duino/STM32Ethernet #71: Ethernet broken since core 2.5.0 for STM32F746G-DISCOVERY

patricklaf, 2023-07-07:
Since core 2.5.0, Ethernet is broken for STM32F746G-DISCOVERY.

Symptoms:
- Ping loss is high, 75-100%.
- NDPClient.begin() never returns.

Core version:
- 2.4.0 is OK.
- 2.5.0 doesn't work.
- 2.6.0 doesn't work.

Libraries:
- STM32Ethernet version 1.3.0.
- LwIP version 2.1.2.
- [NTPClient](https://github.com/arduino-libraries/NTPClient) version 3.2.1.

patricklaf, 2023-07-10:
I have done more tests today and the problem is not in the Ethernet library.
If I enable only Ethernet and NTPClient, it works well with core 2.6.0.

I have to track down what in my application causes this bug since core 2.5.0.
Pretty complicated with lots of libraries and lots of peripherals.

Sorry for the inconvenience.

closed 2023-07-10 by patricklaf

fpistm, 2023-07-10:
Hi @patricklaf 
No worry. Thanks for the feedback.

patricklaf, 2023-07-10:
Hi @fpistm 
Just to add some clue, it seems to be linked to -Og settings.
The whole application compiled with -O0 works perfectly.
If I enable debug, the problem appears.

I don't know if something changed since 2.5.0 regarding debug settings?
BTW it seems to be core related, no Ethernet.