stm32duino/STM32Ethernet #73: Consider adding method end() and operator bool() in EthernetServer

JAndrassy, 2023-09-28:
I did a research on how is Arduino networking API implemented in libraries and I see your library is missing useful functions in the Server class.
https://github.com/JAndrassy/Arduino-Networking-API/blob/main/ArduinoNetAPILibs.md#server-class

fpistm, 2023-09-28:
Hi @JAndrassy 
Looking at official API documentation, I didn't see `end()`. 
https://www.arduino.cc/reference/en/libraries/ethernet/

Only bool:
https://www.arduino.cc/reference/en/libraries/ethernet/ifserver/

Feel free to provide a PR.

JAndrassy, 2023-09-28:
> Feel free to provide a PR.

sorry no.. there are too many libraries with different issues. I don't have a hardware for this one.
(most of the [PR ] links in the linked document are my PR and still many to do)

Arduioi used end() for the first time in official library in the new WiFiS3 library for the Uno R4 WiFi. that library doesn't have a doc yet. but many other networking libraries have end(). users ask for it for temporarily started servers.

fpistm, 2023-11-09:
Hi @JAndrassy 
I've made the implementation in #77

PR #77 (2023-11-09): feat(server): add bool operator and end() api
* bool operator:
https://www.arduino.cc/reference/en/libraries/ethernet/ifserver/

* `end()`:
Not officially documented. Pay attention, when `end()` is called, the socket is not close immediately. The tcp connection will come into `TCP_WAIT`,  after few minutes (~2 minutes) the connection will close completely and free the socket.
So calling `begin()` during this time frame after the `end() ` will silently failed and have to check thanks the bool operator if the server is ready or not. If not then call again `begin()`.

For `TCP_WAIT` ref: https:// [... cut]
commit 6b9664ba (2023-11-09): feat(server): add bool operator

Fixes #73.
commit 88030146 (2023-11-09): feat(server): add end()

Fixes #73.