stm32duino/STM32Ethernet #81: Ethernet.MACAddress should be a getter, not setter

JAndrassy, 2023-12-24:
Ethernet.MACAddress should be a getter, not setter

https://github.com/arduino-libraries/Ethernet/blob/39103da0e1bc569023625ee4693272773397dbb6/src/Ethernet.cpp#L147-L153
```
void EthernetClass::MACAddress(uint8_t *mac_address)
{
	SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
	W5100.getMACAddress(mac_address);
	SPI.endTransaction();
}
```

sstaub, 2023-12-24:
I'm confused, what is the problem for this library? You mention the Arduino Ethernet library which uses the WIZnet chips.

JAndrassy, 2023-12-24:
in this library it is a setter. that is wrong

sstaub, 2023-12-24:
Have a look to the documentation. The implementation is different to the original Arduino library.

fpistm, 2023-12-24:
Agreed with @sstaub . Getter is available with different prototype which is more clear: 
https://github.com/stm32duino/STM32Ethernet/blob/cd50f45d53fefd1b947b2a2552f672f328321a3f/src/STM32Ethernet.cpp#L136C1-L136C1

PR #89 (2024-09-11): fix: align MAC address api with Arduino reference
References:
https://www.arduino.cc/reference/en/libraries/ethernet/ethernet.macaddress/ https://www.arduino.cc/reference/en/libraries/ethernet/ethernet.setmacaddress/

> [!Important]
> Pay attention that `setMACAddress()` have to be called before `Begin()`.
> Else new MAC will be ignored.

This PR also fixes with the default MAC address usage. It is used only if one or more `MAC_ADDRx` defined.
The MAC address is also correct at all level.

Fixes #81

## Test sketch
```C++
#include <LwIP.h>
#include <STM32Ethernet.h>

byte mac[] = { 0xDE, 0xAD, [... cut]
commit 348c968d (2024-09-11): fix: align MAC address api with Arduino reference

References:
https://www.arduino.cc/reference/en/libraries/ethernet/ethernet.macaddress/
https://www.arduino.cc/reference/en/libraries/ethernet/ethernet.setmacaddress/

Pay attention that setMACAddress have to be called before Begin().
Else new MAC will be ignored.

Fixes #81
closed 2024-09-11 by fpistm