stm32duino/STM32Ethernet #83: possible ARP problem

JAndrassy, 2023-12-27:
I can't access a server on Arduino with this library. The library doesn't respond to ARP requests so my computer (Linux) can't route to it. ping doesn't work too. But maybe it is just my weird setup. Can somebody please test a server example?

The truth is I don't have a STM32 MCU with Ethernet peripheral. I test with a Black Pill (F411) with a W5500 module on SPI. I replaced ethernetif.cpp implementation with implementation using W5500 driver from the esp8266 Arduino core. But that should not change how LwIP works. DHCP (UDP) and EthernetClient work.

I ported the QNEthernet library for Teensy 4 to Black Pill with a W5500 to test it because I don't have a Teensy. With QNEthernet library the same Black Pill responds to ARP requests. QNEthernet uses LwIP too.

Expected output in WireShark
```
6	5.180458375	EliteGro_18:13:7f	de:ad:be:ef:fe:ed	ARP	42	Who has 10.42.0.242? Tell 10.42.0.1
7	5.181755916	de:ad:be:ef:fe:ed	EliteGro_18:13:7f	ARP	60	10.42.0.242 is at de:ad:be:ef:fe:ed
```

fpistm, 2023-12-30:
This should ne a discussion more than an issue. Mainly with modifications you made.

JAndrassy, 2024-01-03:
@fpistm can you please check the WebServer example on a board compatible with STM32Ethernet library?

fpistm, 2024-01-03:
@JAndrassy 
I've tested with a DISCO F746NG and it works:

```
10:23:40.884 -> GET / HTTP/1.1
10:23:40.917 -> Host: 192.168.1.177
10:23:40.917 -> Connection: keep-alive
10:23:40.949 -> Cache-Control: max-age=0
10:23:40.981 -> Upgrade-Insecure-Requests: 1
10:23:41.013 -> User-Agent: Mozilla/5.0 (Linux; Android 10; K) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/120.0.0.0 Mobile Safari/537.36
10:23:41.141 -> Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.7
10:23:41.301 -> Referer: http://192.168.1.177/
10:23:41.332 -> Accept-Encoding: gzip, deflate
10:23:41.365 -> Accept-Language: fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7
10:23:41.397 -> 
10:23:41.428 -> client disconnected
10:23:41.492 -> new client
10:23:41.525 -> GET /favicon.ico HTTP/1.1
10:23:41.525 -> Host: 192.168.1.177
10:23:41.557 -> Connection: keep-alive
10:23:41.588 -> User-Agent: Mozilla/5.0 (Linux; Android 10; K) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/120.0.0.0 Mobile Safari/537.36
10:23:41.717 -> Accept: image/avif,image/webp,image/apng,image/svg+xml,image/*,*/*;q=0.8
10:23:41.782 -> Referer: http://192.168.1.177/
10:23:41.814 -> Accept-Encoding: gzip, deflate
10:23:41.846 -> Accept-Language: fr-FR,fr;q=0.9,en-US;q=0.8,en;q=0.7
10:23:41.911 -> 
10:23:41.911 -> client disconnected
```
[image]

closed 2024-01-03 by fpistm