stm32duino/STM32Ethernet #93: EthernetClient BUG

q1098401179, 2025-02-07:
**Please, before reporting any issue**
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](https://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the repository.
- **Avoid** to submit a GitHub issue for project troubleshooting.

Any questions/feedback/suggestions should be discussed on the [stm32duino forum](https://stm32duino.com):
  * questions on the [STM32 Core](https://www.stm32duino.com/viewforum.php?f=35)
  * bugs/enhancements on the [STM core: Bugs and enhancements](https://www.stm32duino.com/viewforum.php?f=38)

&#x26A0; When reporting any issue, please try to provide all relevant information to help on its resolution.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
1. Plug
2. Press USER_BTN
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: [e.g. 1.8.8]
 - STM32 core version: [e.g. 1.4.0]
 - Library version: [e.g. 1.2.0]
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: [e.g. SWD]

**Hardware (please complete the following information):**
 - Board Name: [e.g. Nucleo F429ZI]
 - Hardware Revision: [e.g. Rev B]
 - Extra hardware used if any: [e.g. X-Nucleo IKS01A1]

**Additional context**
Add any other context about the problem here.

q1098401179, 2025-02-07:
void EthernetClient::stop()
{
  if (_tcp_client == NULL) {
    return;
  }

  // close tcp connection if not closed yet
  if (status() != TCP_CLOSING) {
    tcp_connection_close(_tcp_client->pcb, _tcp_client);
  }
  mem_free(_tcp_client);
  _tcp_client= NULL;
}

In this function, memefree (_tcp_cient);, After release, the _tcp_cient was not set to NULL, which caused an exception when TCPClient was reconnected. After adding it, it worked normally. Please fix this issue

fpistm, 2025-02-07:
Hi @q1098401179 
feel free to provide a PR it's a community project.

closed 2025-02-07 by fpistm